mutability/dump1090 #204: Please improve the README regarding the external webserver

E3V3A, 2017-06-30:
I've tried to follow the installation instructions in the README about using an external web-server, in my case `lighttpd` and/or *apache* as mentioned in #203 . Both methods is a no go. 

And navigating to the page: `file:///home/xxxx/dump1090/public_html/gmap.html` give me:
```
Problem fetching data from dump1090.
AJAX call failed (error: NS_ERROR_DOM_BAD_URI: Access to restricted URI denied). Maybe dump1090 is no longer running?
The displayed map data will be out of date.  
```
What does that mean? 

I'm running this on Kali where Apache was already installed and using and RTL SDR with:
```bash
./dump1090 --ppm 1 --gain 30 --interactive --metric --net
``` 
This gives results in CLI but nothing in browser.

BTW. Are there additional steps needed to be taken after: `dpkg-buildpackage -b` as it doesn't seem to have been installed?

mutability, 2017-07-01:
Yes, you need to install the package with dpkg after building it.

The webpage won't work if loaded direct from the filesystem because of browser security restrictions.

E3V3A, 2017-07-02:
* What is the command line you use to start it? 
* What are the configuration files you use?

I'm using: `dump1090-mutability --ppm 1 --gain 30 --interactive --metric --net`
with:

```bash
# cat /etc/lighttpd/lighttpd.conf

server.modules = (
	"mod_access",
	"mod_alias",
	"mod_compress",
 	"mod_redirect",
)

server.document-root        = "/var/www/html"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

# default listening port for IPv6 falls back to the IPv4 port
include_shell "/usr/share/lighttpd/use-ipv6.pl " + server.port
include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"
```
and 

```bash
# cat /etc/default/dump1090-mutability

# dump1090-mutability configuration file
# This is a POSIX shell fragment.
# You can edit this file directly, or use
# "dpkg-reconfigure dump1090-mutability"

# Set to "yes" to start dump1090 on boot.
START_DUMP1090="no"

# User to run dump1090 as.
DUMP1090_USER="dump1090"

# Logfile to log to
LOGFILE="/var/log/dump1090-mutability.log"

#
# Receiver options
#

# RTLSDR device index or serial number to use
# If set to "none", dump1090 will be started in --net-only mode
DEVICE=""

# RTLSDR gain in dB.
# If set to "max" (the default) the maximum supported gain is used.
# If set to "agc", the tuner AGC is used to set the gain.
GAIN="max"

# RTLSDR frequency correction in PPM
PPM="0"

#
# Decoding options
#

# If yes, fixes messages with correctable CRC errors.
FIX_CRC="yes"

# If set, supplies a reference location for local position decoding.
LAT=""
LON=""

# If set, provides the absolute maximum receiver range used to
# filter bad position reports, and to determine when local position
# decoding is safe to use. Specify this in nautical miles (NM).
MAX_RANGE="300"

#
# Networking options
#

# Port to listen on for raw (AVR-format) input connections. 0 disables.
RAW_INPUT_PORT="30001"

# Port to listen on for raw (AVR-format) output connections. 0 disables.
RAW_OUTPUT_PORT="30002"

# Port to listen on for SBS-format output connections. 0 disables.
SBS_OUTPUT_PORT="30003"

# Port to listen on for Beast-format input connections. 0 disables.
BEAST_INPUT_PORT="30004,30104"

# Port to listen on for Beast-format output connections. 0 disables.
BEAST_OUTPUT_PORT="30005"

# TCP heartbeat interval in seconds. 0 disables.
NET_HEARTBEAT="60"

# Minimum output buffer size per write, in bytes.
NET_OUTPUT_SIZE="500"

# Maximum buffering time before writing, in seconds.
NET_OUTPUT_INTERVAL="1"

# TCP buffer size, in bytes
NET_BUFFER="262144"

# Bind ports on a particular address. If unset, binds to all interfaces.
# This defaults to binding to localhost. If you need to allow remote
# connections, change this.
NET_BIND_ADDRESS="127.0.0.1"

#
# Misc options
#

# Interval (in seconds) between logging stats to the logfile. 0 disables.
STATS_INTERVAL="3600"

# Path to write json state to (for use with an external webserver). Blank disables.
JSON_DIR="/run/dump1090-mutability"

# Interval between writing json state (in seconds). 0 disables.
JSON_INTERVAL="1"

# Accuracy of receiver location to write to json state, one of "exact" / "approximate" / "none"
JSON_LOCATION_ACCURACY="approximate"

# Set to yes to log all decoded messages
# This can get large fast!
LOG_DECODED_MESSAGES="no"

# Additional options that are passed to the Daemon.
EXTRA_ARGS=""
```
and:

```bash
# cat /etc/lighttpd/conf-available/89-dump1090.conf

# Allows access to the static files that provide the dump1090 map view,
# and also to the dynamically-generated json parts that contain aircraft
# data and are periodically written by the dump1090 daemon.

url.redirect += (
  "^/dump1090/$" => "/dump1090/gmap.html",
  "^/dump1090$" => "/dump1090/gmap.html"
)

alias.url += (
  "/dump1090/data/" => "/run/dump1090-mutability/",
  "/dump1090/" => "/usr/share/dump1090-mutability/html/"
)

# The stat cache must be disabled, as aircraft.json changes
# frequently and lighttpd's stat cache often ends up with the
# wrong content length.
server.stat-cache-engine    = "disable"
```

and I still get the following in browser:  
http://localhost/dump1090/gmap.html

```text
Problem fetching data from dump1090.
AJAX call failed (error: Not Found). Maybe dump1090 is no longer running?
The displayed map data will be out of date. 
```

mutability, 2017-07-02:
You need a suitable --write-json arg

E3V3A, 2017-07-02:
ok, so what should it be?

mutability, 2017-07-02:
You need to point it at a directory that your webserver then makes available under the data/ subpath.
I suggest putting it on tmpfs as it'll be rewritten once a second.

mutability, 2017-07-02:
In the above config, this is `/run/dump1090-mutability/`

This is, incidentally, the default config if you do a package install (and, in fact, is in the `/etc/default/dump1090-mutability` file above too, so I'm not sure quite what you broke such that the init script is not passing the right arg. If you're starting it by hand.. don't do that. Or at least get the command line right for what you're doing.

E3V3A, 2017-07-03:
ok, I have no idea what's wrong. There is nothing in: `/run/dump1090-mutability/`. In fact it doesn't even exists. It seem we need root permissions to access or write to those locations? I'd love to see a step-by-step instruction how to get from dpkg installation, to the web-page using lighttpd.  It's particualrly annoying as the interactive part seem to work fine, but the AJAX part doesn't get anything. 

* So how to get the server pages working when using CLI command as I show above?

* So how to get the server pages working when using the `sudo service dump1090-mutability start` (and not CLI)?

Are they the same or mutually exclusive?
How can I debug this?

mutability, 2017-07-03:
Since you're running on a distribution I don't use I think you'll have to diagnose this one yourself. As I said above what needs to happen is for dump1090 to be invoked with a --write-json arg that points to an (existing, writable) directory that is then served by the webserver.

closed 2017-07-03 by mutability

E3V3A, 2017-07-03:
I solved it! It now works with this: 

```bash
# cat /etc/default/dump1090-mutability | sed -e 's/^#.*//g' | grep -E "[[:print:]]"

START_DUMP1090="yes"
DUMP1090_USER="xxxx"
LOGFILE="/var/log/dump1090-mutability.log"
DEVICE=""
GAIN="max"
PPM="1"
FIX_CRC="yes"
LAT="40.0"
LON="20.0"
MAX_RANGE="300"
RAW_INPUT_PORT="30001"
RAW_OUTPUT_PORT="30002"
SBS_OUTPUT_PORT="30003"
BEAST_INPUT_PORT="30004,30104"
BEAST_OUTPUT_PORT="30005"
NET_HEARTBEAT="60"
NET_OUTPUT_SIZE="100"
NET_OUTPUT_INTERVAL="1"
NET_BUFFER="262144"
NET_BIND_ADDRESS="127.0.0.1"
STATS_INTERVAL="3600"
JSON_DIR="/run/dump1090-mutability"
JSON_INTERVAL="1"
JSON_LOCATION_ACCURACY="approximate"
LOG_DECODED_MESSAGES="no"
EXTRA_ARGS=""
```

The problem seem to be that just editing the above file and `force-reload` the *lighttpd* service was **not** enough. It seem that you **must** run `dpkg-reconfigure`. Since I'm running in *localhost* I also decreased the minimum amount of packets (NET_OUTPUT_SIZE). In addition it seem that you need to specify yourself as the user to run as, not `dump1090`, but I haven't checked.

The following steps did it. 
```bash
sudo service lighttpd stop
sudo service dump1090-mutability stop
sudo dpkg-reconfigure dump1090-mutability

# Start dump1090 automatically?  <yes>    # as init
# User to run dump1090 as:       xxxx     # default: [dump1090]
# ...

sudo service lighttpd force-reload
sudo service --status-all |grep "\+"
sudo service dump1090-mutability start
```

Please add this info to your README, because this was not at all obvious from reading it. 
Apart from that, thank you very much for your great package and quick response. 

:)

mutability, 2017-07-03:
It's strange, because what dpkg-reconfigure will do is:

 * stop dump1090-mutability
 * update /etc/default/dump1090-mutability from your answers
 * enable the lighttpd config fragment if you said to enable it
 * restart lighttpd
 * start dump1090-mutability

... essentially all the same steps that are done on package install.