mutability/dump1090 #205: About Interrogator's identity in DF 11

billdimitrakis, 2017-07-03:
Hello,

I was checking the output of dump1090 and I noticed that in packets with DF format 11, there are values for Interrogation identity that exceed the value of 80. According to  https://opensky-network.org/dasc2016/interrogators.html and the Mode -S specification, the Interrogator's identity takes values up to 80. More specifically, the first 17 bits of PI field are zeros , the next three correspond to CL field and the last four to the SI field. However, according to specification, CL field can only take values up  to 4, otherwise these values are not valid. In the recording that I took,  I have e.g. IID 102-> 110 0110. So CL =6, however, this is not valid according to specification. 
Am I missing something or is it a bug of the decoder?

Kind regards,
Vasilis

mutability, 2017-07-04:
Do you have a sample message?

mutability, 2017-07-04:
IID 102 just means "the CRC residual was 102"; since it can't see the interrogations, dump1090 has no way to know if that's really an IID or if it's a damaged message. (It generally trusts messages with IID != 0 less for this reason).

If there are some values of IID that are invalid, then that could be used to discard a few more bad messages.

billdimitrakis, 2017-07-04:
Thanks a lot for your reply, I also checked now that it is the CRC residual and I understood what you mean

closed 2017-08-19 by mutability