mutability/dump1090 #210: How to get a selected aircraft in a json-file

audi2010, 2017-09-14:
I user Dump1090-mutability with a Raspi3 and NooElec NESDR Nano 3.
It works great.
Now I want to save the browser-selected aircraft in a file "selected.json" in /run/...
Script.js on Row 607 shows "selected = Planes[SelectedPlane]"
Is there a way to change this script, so save the selected plane and update, if the function is called?

mutability, 2017-09-14:
No, because the browser has no way to write files on the Pi.

closed 2017-09-14 by mutability

audi2010, 2017-09-14:
Thank you, for your very rapid answer. 
My hope was, to get this date, to controll 2 servos with a camera, controled by a wemos mini.

mutability, 2017-09-14:
You might want to set up an ajax endpoint that the browser can POST to when the selection changes and capture it that way

audi2010, 2017-10-02:
After two weeks searching and trying an other way to get position-data from aircraft.json over php, 
My Problem is, I don't know, how to setup an ajax endpoint. Is there an easy way, to do this with my miserable knowledge ?