mutability/dump1090 #211: undefined symbol: rtlsdr_set_bias_tee when building with BIAS-T support

gleven, 2017-09-30:
I tried to build the package for BIAS-T but dump1090 will not start. 
The log gives me the following error
**/usr/bin/dump1090-mutability: symbol lookup error: /usr/bin/dump1090-mutability: undefined symbol: rtlsdr_set_bias_tee**

Am i missing something?

mutability, 2017-09-30:
Sounds like you have linked against a different library compared to the library available at runtime.

gleven, 2017-09-30:
If you mean an rtlsdr library, i only have one version installed as i did a fresh install to make sure that i don't have such issues.
Is there a step-by-step guide so i can check if i did something wrong in the procedure?

mutability, 2017-09-30:
You'd probably have to contact the original author of the PR that added bias-tee support, I have no experience with it myself.

gleven, 2017-10-01:
Seems like the problem is caused by librtlsdr-dev as it also installs librtlsdr0.
librtlsdr-dev is required by dump1090.
How can i bypass this?

gleven, 2017-10-27:
I finally managed to install it with bias-t. Closing issue

closed 2017-10-27 by gleven