mutability/dump1090 #218: Surface position ground speed calculation

jreuning, 2018-01-23:
I'm curious about the ground speed calculation for ES surface position messages.  Based on the ICAO publication, it seems like the logic is off by one step for values 8-123, and missing for 2-7. Or did I misread it?

I tracked the code to a commit in the MalcolmRobb fork but thought maybe someone here would know.

https://github.com/mutability/dump1090/blob/cc53c90837520bcf1764ce051c794d1c7f8de084/mode_s.c#L182-L199

mutability, 2018-01-23:
It returns the upper bound of the speed range, truncated to an integer.

Encoded values 2-7 all represent ranges with an upper bound less than 1kt, so they're all 0.
The other encoded values should be OK if you interpret them as the upper bound (e.g. encoded value 39 is 15kt < speed <= 16kt and returns 16)

I'm in the process of changing this over to floating point, returning the midpoint of the range, & handing ADS-B v2 (which changes some of the encodings) over in the flightaware fork.

jreuning, 2018-01-23:
Okay, interesting.  I read the docs as the upper bound being exclusive. The table I found shows it as 15 kt <= speed < 16 kt.  I don't suppose there's a public doc that shows how v2 differs?

mutability, 2018-01-23:
v2 changes the upper bound to be inclusive and changes the encoding for low speeds. It's specified in DO-260B / ED-102A, which you'd need to pay for (maybe you can find a draft online)

jreuning, 2018-01-23:
@mutability thank you for quick response and extra detail on this! 👍

closed 2018-01-23 by jreuning