mutability/dump1090 #220: Fails to compile on Ubuntu 17.10

older-pack, 2018-02-02:
I'm very new to compiling (my very first time) and this isn't compiling for me.
Can any one help me out?
Its failing with these messages:
```interactive.c: In function ‘interactiveShowData’:
interactive.c:143:44: error: ‘%03d’ directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                            ^~~~
interactive.c:143:42: note: directive argument in the range [-21474836, 21474836]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                          ^~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from dump1090.h:65,
                 from interactive.c:50:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 5 and 11 bytes into a destination of size 6
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:49: recipe for target 'interactive.o' failed
make[2]: *** [interactive.o] Error 1
make[2]: Leaving directory '/home/emoh/Downloads/dump1090/mutability/dump1090-master'
dh_auto_build: make -j1 "EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090-mutability/html\\\" -DMODES_DUMP1090_VARIANT=\\\"dump1090-mutability\\\"" returned exit code 2
debian/rules:21: recipe for target 'override_dh_auto_build' failed
make[1]: *** [override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/emoh/Downloads/dump1090/mutability/dump1090-master'
debian/rules:30: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
```

Thanks.

older-pack, 2018-02-08:
Anyone any ideas at all? ...please :)

mfvescovi, 2018-02-08:
Have a look at [this](https://salsa.debian.org/debian-hamradio-team/dump1090-mutability/blob/master/debian/patches/0003-gcc7.patch).

older-pack, 2018-02-08:
Fantastic. Thanks mfvescovi. 

Being a real beginner, and not using git, I simply copied and pasted your edits into my local copy of 'interactive.c'.
Then I ran `dpkg-buildpackage -b` and it worked! 
Out popped `dump1090-mutability_1.15~dev_amd64.deb`

So I installed that with `sudo dpkg -i dump1090-mutability_1.15~dev_amd64.deb`

And it runs with:
`$ dump1090-mutability`

Its worth noting I'm not using ver 1.14. I am using version:
	`Latest commit cc53c90 Jan 10, 2018`
	https://github.com/mutability/dump1090/commit/cc53c90837520bcf1764ce051c794d1c7f8de084

Which is called v1.15~dev when compiled.
But your v1.14 patches still seem to work with it.

I haven't managed to get the web part of it, with the maps, running yet though.
I can't figure out how to get the built in HTTP server working.

older-pack, 2018-02-08:
Here is my patched 'interactive.c' for anyone else who could use it.
https://gist.github.com/older-pack/b2781da9549d447a6f347a2107be4268

older-pack, 2018-02-10:
Finally got it all going. Couple of points I noted for any other Ubuntu users:

- Ubuntu's lighttpd web server has a [bug](https://bugs.launchpad.net/ubuntu/+source/lighttpd/+bug/1453463). You need to install a package called 'gamin' to fix it.
- For me, there seemed to be a permission issue with the dump1090's default json folder. No matter what I did, no matter how I reconfigure it, no data was ever written to it. So I had to change it to a non-root one. That was the big breakthrough for me. I chose `/var/tmp/dump1090-mutability` (as well as chnaging it for dump1090 you also have to change it for lighttpd in `/etc/lighttpd/conf-available/89-dump1090.conf`.) 
- The configuration wizard has a much longer run through if you choose start dump1090 automatically. If you choose not to, then the wizard ends abruptly  (Maybe a bug? as the default json folder doens't get created if you chose not auto start.)
- I don't want it running automatically (as a service/daemon). I just want to run it from the command line when I want to. But by setting it up to run automatically I was able learn some stuff. Then I reconfigured it to not run automatically when I was done. 
- I copied the command line that the automatic setup uses and I use a slightly modified version now:
  `$ dump1090-mutability --net --ppm 0 --fix --lat 64.135 --lon -21.865 --max-range 300 --net-bind-address 127.0.0.1 --stats-every 3600 --write-json /var/tmp/dump1090-mutability --write-json-every 1 --json-location-accuracy 1 --interactive`
  or a simpler version `$ dump1090-mutability --interactive --net --write-json /var/tmp/dump1090-mutability`
  And to view in the web browser the URL is `http://127.0.0.1/dump1090/gmap.html`

These were helpful
https://github.com/mutability/dump1090/issues/204
https://forum.flightradar24.com/threads/10232-How-to-Install-dump1090-mutability_1-15-dev-on-RPi

closed 2018-02-10 by older-pack