mutability/dump1090 #222: Building dump1090-mutability 1.15: error dpkg-shlibdeps: error: no dependency information found for /usr/lib/librtlsdr.so.0

Flint4, 2018-02-12:
Hi, I am facing an issue for which I can't find a solution by myself. I am running the latest Raspian Stretch release on my RPi2. I tried to buld dump1090-mutability v1.15-dev with

```
apt-get install librtlsdr-dev libusb-1.0-0-dev pkg-config debhelper git-core

wget https://github.com/mutability/librtlsdr/releases/download/v0.5.4_git-1/librtlsdr0_0.5.4.git-1_armhf.deb

wget https://github.com/mutability/librtlsdr/releases/download/v0.5.4_git-1/librtlsdr-dev_0.5.4.git-1_armhf.deb

dpkg -i librtlsdr0_0.5.4.git-1_armhf.deb

dpkg -i librtlsdr-dev_0.5.4.git-1_armhf.deb

git clone https://github.com/mutability/dump1090.git

cd dump1090

dpkg-buildpackage -b
```
But building aborts with the error

```
dh_shlibdeps
        dpkg-shlibdeps -Tdebian/dump1090-mutability.substvars debian/dump1090-mutability/usr/bin/dump1090-mutability debian/dump1090-mutability/usr/bin/view1090-mutability

dpkg-shlibdeps: error: no dependency information found for /usr/lib/librtlsdr.so.0 (used by debian/dump1090-mutability/usr/bin/dump1090-mutability)
Hint: check if the library actually comes from a package.
```

I could not find any working solution with Google, that's why I post it here. Any ideas what to do? Is it related to Stretch? 

Brgds,
Alex

mutability, 2018-02-12:
This generally means that it's linking to a non-package-installed version of librtlsdr, not the package-installed version

Flint4, 2018-02-13:
Thank you for the hint. I purged librtlsdr already multiple times and installed the version mentioned above. If I understand you correctly I have to look after remainings of librtlsdr after purge and kill them?

mutability, 2018-02-13:
You need to remove the path that dpkg-shlibdeps mentions, which is a copy of the library which was manually installed at some point (not managed by a package)

e.g. on my (x86) system here, the librtlsdr0 package installs `/usr/lib/x86_64-linux-gnu/librtlsdr.so.0.5.3` and `/usr/lib/x86_64-linux-gnu/librtlsdr.so.0` only; it does not install `/usr/lib/librtlsdr.so.0`. You can check which files are managed by the package by `dpkg -L librtlsdr0`

Flint4, 2018-02-14:
> it does not install /usr/lib/librtlsdr.so.0

Thank you so much! I checked the package contents and it was like you are saying. After I removed `/usr/lib/librtlsdr.so.0` it works! 

Thank you again for the help and for everything you are doing for the community! 

Alex

closed 2018-02-14 by Flint4