mutability/dump1090 #224: Fails to build on Ubuntu 17.10. (gcc 7.2.0)

jprochazka, 2018-03-09:
Building dump1090 on Ubuntu 17.10 results in a failed build "...directive output may be truncated..." with the following compiler output:

**Note:**
_(dump1090-fa does not appear to have this issue and builds fine)_
```
dpkg-buildpackage: info: source package dump1090-mutability
dpkg-buildpackage: info: source version 1.15~dev
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by Oliver Jowett <oliver@mutability.co.uk>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build dump1090
 fakeroot debian/rules clean
dh clean
dh: Compatibility levels before 9 are deprecated (level 8 in use)
   dh_auto_clean
dh_auto_clean: Compatibility levels before 9 are deprecated (level 8 in use)
        make -j1 clean
make[1]: Entering directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
rm -f *.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o dump1090 view1090 faup1090 cprtests crctests
make[1]: Leaving directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
   dh_clean
        rm -f debian/debhelper-build-stamp
        rm -rf debian/.debhelper/
dh_clean: Compatibility levels before 9 are deprecated (level 8 in use)
        rm -f -- debian/dump1090-mutability.substvars debian/files
        rm -fr -- debian/dump1090-mutability/ debian/tmp/
        find .  \( \( \
                \( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS \) -prune -o -type f -a \
                \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
                 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
                 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
                 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
                \) -exec rm -f {} + \) -o \
                \( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
 debian/rules build
dh build
dh: Compatibility levels before 9 are deprecated (level 8 in use)
   dh_update_autotools_config
   dh_auto_configure
dh_auto_configure: Compatibility levels before 9 are deprecated (level 8 in use)
   debian/rules override_dh_auto_build
make[1]: Entering directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
dh_auto_build -- 'EXTRACFLAGS=-DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\"'
dh_auto_build: Compatibility levels before 9 are deprecated (level 8 in use)
        make -j1 "EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090-mutability/html\\\" -DMODES_DUMP1090_VARIANT=\\\"dump1090-mutability\\\""
make[2]: Entering directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
gcc -Wdate-time -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fdebug-prefix-map=/home/joe/adsb-receiver/build/dump1090-mutability/dump1090=. -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE `pkg-config --cflags librtlsdr libusb-1.0` -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c dump1090.c -o dump1090.o
gcc -Wdate-time -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fdebug-prefix-map=/home/joe/adsb-receiver/build/dump1090-mutability/dump1090=. -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c anet.c -o anet.o
gcc -Wdate-time -D_FORTIFY_SOURCE=2 -DMODES_DUMP1090_VERSION=\"v1.15~dev\" -g -O2 -fdebug-prefix-map=/home/joe/adsb-receiver/build/dump1090-mutability/dump1090=. -fstack-protector-strong -Wformat -Werror=format-security -O2 -g -Wall -Werror -W -std=c11 -D_DEFAULT_SOURCE -DHTMLPATH=\"/usr/share/dump1090-mutability/html\" -DMODES_DUMP1090_VARIANT=\"dump1090-mutability\" -c interactive.c -o interactive.o
interactive.c: In function ‘interactiveShowData’:
interactive.c:143:44: error: ‘%03d’ directive output may be truncated writing between 3 and 9 bytes into a region of size 5 [-Werror=format-truncation=]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                            ^~~~
interactive.c:143:42: note: directive argument in the range [-21474836, 21474836]
                         snprintf(strFl,6,"F%03d",((a->altitude+50)/100));
                                          ^~~~~~~
In file included from /usr/include/stdio.h:862:0,
                 from dump1090.h:65,
                 from interactive.c:50:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 5 and 11 bytes into a destination of size 6
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:49: recipe for target 'interactive.o' failed
make[2]: *** [interactive.o] Error 1
make[2]: Leaving directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
dh_auto_build: make -j1 "EXTRACFLAGS=-DHTMLPATH=\\\"/usr/share/dump1090-mutability/html\\\" -DMODES_DUMP1090_VARIANT=\\\"dump1090-mutability\\\"" returned exit code 2
debian/rules:21: recipe for target 'override_dh_auto_build' failed
make[1]: *** [override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/joe/adsb-receiver/build/dump1090-mutability/dump1090'
debian/rules:30: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2
```
The output for `gcc --version` on Ubuntu 17.10:
```
joe@ubuntu:~$ gcc --version
gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
```

mutability, 2018-03-10:
please reopen if that doesn't fix it (I don't have a gcc 7 system on hand to test on right now)

closed 2018-03-10 by mutability

jprochazka, 2018-03-22:
Works now. Thanks for the fix!

commit f65759d3 (2018-03-10): Disable gcc -Wformat-truncation warning due to false positives.
Untested other than verifying that older gccs don't complain about the option.
Might fix #224, #200, #201