mutability/dump1090 #226: Switch between dump1090 and rtl_tcp

martenumberto, 2018-03-20:
Hello,

usually i run dump1090 on my pi to upload the data to Fr24. But sometimes I want to hear some Radio with rtl_tcp. When I quit dump1090 and run rtl_test everything is fine. But when I run rtl_tcp it quits with an error:
"rtl_tcp: symbol lookup error: rtl_tcp: undefined symbol: rtlsdr_set_bias_tee"

I think the problem is librtlsdr. When I deinstall it and remove the libs from /usr/local/lib, rtl_tcp runs fine again. But now dump1090 is broken because of the missing librtlsdr.

Any ideas to solve the problem? 
In the second step I want to create a script which switches easily between dump1090 and rtl_tcp

marcus-aa, 2018-03-23:
Buy a second dongle. They should have different antennas anyways...

mutability, 2018-03-23:
The link problem is because you have more than one version of librtlsdr installed.

closed 2018-03-23 by mutability

martenumberto, 2018-04-11:
What can I do to solve it? I've bought a second sdr stick and I want to run dump1090 and rtl_tcp

mutability, 2018-04-11:
Pick one version of rtlsdr to use, make sure only that version is installed, and make sure that both dump1090 and rtl_sdr were built against that version