mutability/dump1090 #227: Re-enable --aggressive mode.

marcus-aa, 2018-03-23:
We have very good experience running decoding in low-traffic areas with 

--oversample --fix --phase-enhance --aggressive 

We see no problems with false decodes, and disabling aggressive means a loss of messages.
Several users have confirmed that disabling -aggressive makes their systems perform worse.

Please re-enable it in new versions.

mutability, 2018-03-23:
There's already an #ifdef; I'm not going to enable it by default because I have strong evidence in the past that it _does_ cause garbage data. If you do turn it on, please don't feed your data anywhere.

closed 2018-03-23 by mutability

marcus-aa, 2018-03-23:
I'm not in any way suggesting it to be on as  default just as a usable argument, but it doesn't run at all on 1.15 64-bit;
```
ADSB-RX:/usr/bin$ dump1090-mutability --aggressive
warning: --aggressive not supported in this build, option ignored.
Fri Mar 23 16:54:50 2018 EET  dump1090-mutability v1.15~dev starting up.
```

There is a reference here https://github.com/mutability/dump1090/commit/577fe9b52e248dff70f3c22959b83c958cdacda0 
to it having been removed (but it's quite possible that I'm doing something wrong...)

*I have 50 systems running in this config all feeding my server, I would notice very quickly if I get garbage. Many of them are also feeding FA and FR24 without any problem.*

mutability, 2018-03-23:
You need to rebuild with the #ifdef turned on if you really want --aggressive.

To reiterate: _please do not feed FA with data collected with --aggressive enabled_; we value data quality over data quantity, and I am absolutely sure that --aggressive produces a fairly high rate of garbage (around one message an hour that passes all the sanity checks but is actually garbage, from memory). FR24 may have a similar policy, I don't know.

It's an error rate that might be acceptable for your own personal use, which is why the option is still in there, but be sure you know what you're doing when you turn the #ifdef back on and make sure that the bad data doesn't make it further out. Even low error rates add up fast when you have 15,000 feeders.

One option is to ensure you're running with --net-verbatim, which will produce the original uncorrected message on port 30005 et al, so that software that makes use of that (e.g. faup1090 for piaware) can make its own decisions about it.

mutability, 2018-03-24:
Actually, what could work here is forcing verbatim mode on if --aggressive is in use.
Would that work for your case?

leearmstrong, 2018-03-24:
Lee from https://planefinder.net and I echo @mutability’s comments on the aggressive producing garbage. It does. And a lot!!!

I think the proposed forcing verbatim is a good idea!

marcus-aa, 2018-03-24:
Well, as the network feeds goes straight into VRS, without further CRC checking that is not really an option. We have that problem with the FR24 receivers today, the 30334 feed seems to lack CRC checks. Solved by piping it through beast-splitter, but the dump feeders provide much cleaner data, **with** aggressive....

Does mlat-client perform CRC-checking before using the data? 

I'm still interested in why we see so different result from --aggressive. Maybe here with lower traffic intensity and much less overlapping messages the problem is much smaller? But from so many receivers over 2,5 years I don't see any problems from this. While we are not the size of the commercial networks, we still have millions and millions of messages every day.

mutability, 2018-03-24:
Both mlat-client and faup1090/piaware will only accept messages with correct CRCs; this is the desired behavior for them when given a verbatim feed.

If you want to feed a verbatim feed into VRS, VRS will need to do any CRC correction that you want to apply; I suggest you talk to the VRS author about that if it doesn't support that currently.