mutability/dump1090 #229: FC0012 tuner

Basti77, 2018-03-27:
Hello, I may have found a solution for FC0012 tuner.
The USB stick must first be initialized to a different frequency <980Mhz, then you can switch to the 1090Mhz.
Under windows SDR# it works!
Is there a coder that could install this in dump1090-fa?

mutability, 2018-03-28:
I don't have fc0012 on hand to test with, but a couple of questions:

Does it actually get data at 1090mhz? It's possible that the second tuning
attempt is silently failing

Does this work on more than one unit? I wonder how much variation between
chips there is. The limits for a r820t vary from chip to chip

Finally if this is a somewhat reliable thing then it is probably worth
trying to get it into librtlsdr itself rather than a hack in dump1090

Basti77, 2018-03-28:
I too can reproduce it.
Tuners start at 980 and then switch to 1090.
I can not say if it works wonders. I did not want to buy another one;)

But the stick had a great gain compared to another stick with E4000 tuner.

mutability, 2018-03-28:
I ordered a FC0012 (£6, they have got really cheap!) so I can try this out in a few days

Basti77, 2018-03-29:
Spontaneously made in the office:

The signal is not clean because I took only a piece of wire as an antenna for the screenshot.

At home with a 1090 antenna on the roof it is better.

[image]

Basti77, 2018-03-29:
Another info.
  the stick can do a lot more.
[image]

a further note for this debugging.
I tested another software.
SDR Console.
Since the stick is running without "tricks".

mutability, 2018-03-31:
No luck with the FC0012 I have here unfortunately.

a) tuning to 980MHz fails (rtlsdr complains the PLL parameters are out of range)
b) tuning to 1090MHz fails (same as case a)
c) tuning to 940MHZ doesn't complain

If I tune to 940 then to 1090 then I don't actually hear any Mode S signals.

```
Sat Mar 31 17:11:14 2018 BST  dump1090-fa  starting up.
rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN 00000001)
Found Fitipower FC0012 tuner
rtlsdr: tuner gain set to 19.2 dB
tune to 940MHz..
tune to 1090MHz..
[FC0012] no valid PLL combination found for 1090000000 Hz!
```

Usually librtlsdr bails out when it can't find a suitable PLL setting and doesn't actually retune the chip, so this is not surprising - the tuner is still tuned to 940MHz in this case.

I tried tweaking librtlsdr so that it ignores the PLL-out-of-range problem and goes ahead and writes the changes to the tuner anyway, but that made no difference.

So I guess you should check that you really are tuned to 1090MHz and not some other frequency. (Check if you can actually demodulate mode S data at 1090MHz). If you can get 1090MHz data from SDR# et al, I can only assume they're initializing the tuner differently to what rtlsdr does and I'd need details of what they are doing differently to do anything more.

Basti77, 2018-04-01:
I can only interpret from the spectrum. unfortunately there is no longer the streaming plugin for SRD # to download, or I did not find it.
The question is, what makes the windows driver different?

Here is the signal with my ADS antenna (no wire)
should I record?
[image]

mutability, 2018-04-01:
Yeah if you can get an IQ capture @ 2.4MSPS then I can look at that

Basti77, 2018-04-03:
https://drive.google.com/file/d/1Gu28SvNvdc26gD0AmOyUDVPTkjfByIB5/view?usp=sharing

mutability, 2018-04-03:
I ran this through dump1090 both as-is and after resampling to 2.4MSPS with no luck, though it's possible that I got the format conversion wrong (it seems OK from eyeballing it at least)

Basti77, 2018-04-04:
https://drive.google.com/file/d/10-Hcp5iOjXXI0p3ZUlLDBApN7dh2kaQw/view?usp=sharing

Excuse me.
that was my fault.
I just repeated the recording with 2.4Mb.

Basti77, 2018-04-06:
I just tried to reproduce all with the E4000 stick.
So I see no signal even though he displays with Piaware aircraft. very strange.

mutability, 2018-05-09:
This seems to be unlikely to work with the current information available; rtlsdr doesn't know how to program the FC0012 for 1090MHz as far as I can see.

closed 2018-05-09 by mutability