mutability/dump1090 #230: Eurocontrol Maastricht Upper Area Control (MUAC) Map overlay

PW00X, 2018-04-13:
Needless to say I love dump1090 and like most people especially this fork.

As I'm not to powerful on webcoding I was wondering if someone could guide me to integrate the MUAC charts as a overlay for OpenStreeMap.
These "images", actually they are originally PDF, are available on the [Eurocontrol website](http://www.eurocontrol.int/articles/atc-charts) especially the [Maastricht UAC Sectorisation chart](http://www.eurocontrol.int/sites/default/files/article/content/documents/muac/charts/102-0271_180329.pdf) is interesting to me personally as this is the region my install is active and combined with an airband scanner.

Off topic:
The airband scanner is active on all MUAC frequencies available on the chart and provides the tuned frequency (sector name) as a title (NooElec NESDR SMArTee --> rtl_airband --> Icecast)

You guys probably see where I'm once my additional components have arrived;
current setup for ADS-B: [Albrecht Allband 61700](https://www.amazon.co.uk/Albrecht-61700-Allband-scanner-antenna/dp/B005CN97WW) --> [RTL-SDR v3](https://www.rtl-sdr.com/buy-rtl-sdr-dvb-t-dongles/)
In the future the v3 will be used for airband and an on order FlightAware pro + 1090Mhz filter will replace it for ADS-B reception

Feel free to checkout my [current setup in action](https://whatsup.gijsbrechts.tk/)

Regards,
PW00X

mutability, 2018-05-09:
PDFs are going to be hard to work with. If you can get the maps into a tiled bitmap format, though, then that's fairly easy to add as a new layers.

HNJAMeindersma, 2018-05-23:
Not in any easy way. You would need to convert them like @mutability said, cut them up and then find a way to import them with the OpenStreetMap javascript library (CMIIW).

closed 2019-09-14 by mutability