mutability/dump1090 #241: Feature request: add/include sequence number as top level key in JSON output files

dcj, 2018-12-29:
I am reading/processing each JSON output file of dump1090-fa, and my file-reading code is triggered by an inotify watch/callback.
AFAICT, this is working great, but I'd love to have "proof" that I am indeed capturing each file.
I propose to add a top level key to the JSON output files, which increments by one for each file type.
If this existed, then I could keep track of the last sequence number, compare it to the current one, and know that I hadn't missed one.
I have two alternate proposals for what the initial value (upon startup) of the sequence number should be, either:

- zero, or
- seconds since epoch

In either case, each successive sequence number would be +1 of the previous value.
I am thinking that the initial sequence number based on epoch seconds might be handy to record the startup time by readers, if they care to know. But I don't have a strong preference either way.
Thoughts????

I'd really love to have this feature.  In the spirt of open-source, I'll either find a way to write this myself (or find and compensate someone to do this for me), and if so, I'll make a pull request.  
If this were to happen, would you consider merging in this feature (assuming it met your code-quality requirements)?

mutability, 2018-12-29:
Does the existing "now" field work for this?

dcj, 2018-12-29:
Are "now" values **guaranteed** to increment in a fixed sequential order?
I would guess/assume not.  If so, then now isn't what I need....
I guess to restate the problem I am trying to solve, "How can I tell if I miss an output file by reading (what I hope/assume/expect to be) successive versions of that file?  
AFAICT aircraft.json is moved/renamed from a new temp file every second.
If I am reading what I expect to be successive instantiations of that file, how can I know for certain I haven't missed one?  
OTOH, if "now" is always one second later each time, then that might work...

mutability, 2018-12-30:
`now` is monotonically increasing (within a single dump1090 run, at least); it reflects the system time when the file was generated, and system time is also used to decide when to generate a new file.

`refresh` in `receiver.json` tells you the expected interval between updates. You can use this to control your polling interval (probably irrelevant if you're using inotify though). You could use this to detect missed updates e.g. look for an interval that is > 2x the refresh value (small chance of false positives if for some reason dump1090 is very late in generating the file, but you'll have other problems if dump1090 is getting that delayed)

closed 2019-09-14 by mutability